ghi-electronics/Docs #83: Provide a small example to access the file system

Arke64, 2017-12-06:


Arke64, 2017-12-11:
Initial code from Viktor, still in progress.

[SDreader.zip](https://github.com/ghi-electronics/Docs/files/1548373/SDreader.zip)

VictorDuma, 2017-12-27:
Add functions:
ReadDirectory
WriteToFile
AddFile - cant test.  reason data changed from spi
[SDreader.zip](https://github.com/ghi-electronics/Docs/files/1589866/SDreader.zip)

Arke64, 2018-02-19:
Give an example using something like http://old.ghielectronics.com/community/codeshare/entry/559 from Taylorza, just a simple RAM system to show how to use the API and read/write a file

VictorDuma, 2018-02-21:
A little bit tricky without documentation.
class Program
    {
        
        static void Main()
        {
            byte[] data = { 0, 1, 2, 3, 4, 5 };
            byte[] ReadBuf = new byte[6];

            var blockDriver = new MemoryBlockDriver();
            var tfs = new TinyFileSystem(blockDriver);

            tfs.Format();
            tfs.Mount();

            var stream = tfs.Open("GHI", FileMode.CreateNew, 8);

            stream.Write(data, 0, 6);
            stream.Read(ReadBuf, 0, 6);
  
            var file = tfs.GetFiles();
  
            foreach (var i in ReadBuf)
                Debug.WriteLine(i.ToString());

        }
    }

Arke64, 2018-04-23:
We may add file system itself in 1.0, so this may not be needed.

Arke64, 2018-07-09:
Since we added filesystem itself, not needed.

closed 2018-07-09 by Arke64