ghi-electronics/Docs #219: Life support

gus-ghielec, 2018-11-13:
We need to add a note about not using the product in life support applications.

JoelGHI, 2018-11-13:
Our old disclaimer mentions "life support." Other companies have disclaimers that seem more robust. For example, from https://www.nexperia.com/quality/quality-reliability-disclaimer.html:

Nexperia products are not designed, authorized or warranted to be suitable for use in medical, military, aircraft, space or life support equipment, nor in applications where failure or malfunction of an Nexperia product can reasonably be expected to result in personal injury, death or severe property or environmental damage nor for any other use that is not explicitly agreed between the parties. Nexperia accepts no liability for inclusion and/or use of Nexperia products in such equipment or applications and therefore such inclusion and/or use is at the customer's own risk.

Any thoughts about how far we should go with our disclaimer?

gus-ghielec, 2019-02-06:
Added license and details to all products

closed 2019-02-06 by gus-ghielec